RobTillaart/Arduino #147: download

alexbaekeland, 2020-04-05:
how do I download it? I'am folowing a tutorial of How to Mechatronics. I just don't know how to download this one please help.

RobTillaart, 2020-04-05:
Can you give a link to that description?

RobTillaart, 2020-04-05:
- select the tab code
- go to the right under the red line
- click the green download button
- click the download ZIP button
   ==> the archive is downloaded to your computer

unzip the file and move the libraries needed to your libraries folder 
(see the Arduino IDE -> FILE menu -> PREFERENCES -> sketchbbok location

use sketchbookfolder/libraries to store your own libs.

alexbaekeland, 2020-04-05:
Dear Rob

Thank you so mucht for replying .

Alex

RobTillaart, 2020-04-06:
I assume I can close the issue?

alexbaekeland, 2020-04-06:
Dear Rob

Yes you may close the case, thank you for your help.

Kind regards .
Alex

closed 2020-04-06 by RobTillaart