RobTillaart/Arduino #149: Minor accuracy improvement to running average

bourkemcrobbo, 2020-04-16:
I was just looking through your code and noticed a possible minor improvement that can be made. The getAverage() function creates a new variable to hold the sum. If you change this to instead use the _sum variable and set it to zero before computing the sum, future calls to getfastAverage() will benefit from the updated accurate sum.

RobTillaart, 2020-04-16:
Thx for remark. I will look into this asap.

RobTillaart, 2020-04-16:
Yes the example that checks for the difference now just stays at zero, so it is definitely an improvement. Will push update today.
 
note that I'm working on a separate repo for each library to get them into the library manager.  
So several are already "moved".

RobTillaart, 2020-04-16:
Please verify latest version, if problem not solved, please reopen issue.

closed 2020-04-16 by RobTillaart