RobTillaart/Arduino #152: TWBR not declared as variable

kostashellas, 2020-05-14:
little correction to make the sketch usable for newbies.

TWBR not declared as variable in the sketch.

int TWBR; should be added.

thank you for sharing!!

RobTillaart, 2020-05-14:
Which library are you refering too?
What platform are you using?
Twbr is avr specific so it will not work on e.g. esp32.
Think ithe lib in question needs an update.

kostashellas, 2020-05-14:
MultiSpeedI2CScanner

kostashellas, 2020-05-14:
Hi Rob,
Indeed I'm using esp32 platform, I'm sorry I overlooked your email

Anyway, by adding this as int variable in the sketch working well in esp32

Best Regards,  Kostas

RobTillaart, 2020-05-14:
I will check for a fix this week, 
thanks for pointing out

kostashellas, 2020-05-14:
Thank you for sharing, stay safe
Cheers

RobTillaart, 2020-05-16:
Find attached a prelim 0.1.10 version, please test and let me hear if it works or still fails.
[MultiSpeedI2CScanner.zip](https://github.com/RobTillaart/Arduino/files/4638418/MultiSpeedI2CScanner.zip)

kostashellas, 2020-05-18:
Hi Rob, tested, not working.
see the files attached from both sketches the one that is working and the new provided that does not.
[working.txt](https://github.com/RobTillaart/Arduino/files/4644808/working.txt)
[not working.txt](https://github.com/RobTillaart/Arduino/files/4644809/not.working.txt)

closed 2020-05-18 by kostashellas

RobTillaart, 2020-05-18:
Thanks for testing,

That is a really old version (2014) that is working.

But OK, I need to do my own testing for UNO and ESP32 with the new version to get it working.