RobTillaart/Arduino #163: MultiMap examples use wrong include

guibom, 2020-11-03:
Hello!
I've installed your MultiMap library with the Arduino Library Manager, but none of the examples were compiling.

The issue is that they are using `#include "multimap.h"`, however that's the wrong name and it should be `#include "MultiMap.h"`

Thanks for all your awesome libraries!

RobTillaart, 2020-11-04:
Thanks for pointing to this bug.
What OS do you use?

RobTillaart, 2020-11-04:
The latest version can be found here - https://github.com/RobTillaart/MultiMap
Most of my libraries got their own repository and with updates some things might have been broken.
Best is to move the library from the Arduino library folder (keep a backup) and do a clean install

Let me know if that solves the problem

closed 2020-11-10 by RobTillaart