RobTillaart/Arduino #165: Declaring multiple RAs, possible ?

unstopp, 2020-12-27:
Hi Rob,

Congrats on the library, I find it very useful for my needs so far, thank you!

I ran into an issue, it seems when I try to use 2 RAs variables, the Arduino freezes after adding the value to the second RA (from my experience, might be a buffer issue). Apologies if this is a noob question.

Any thoughts on this ? I need 5 RAs and 5 RMs for my sensors.

Background: I'm using RM and I take that value and add it to RA. Works great with just 1 RA. It seems multiple RMs are working (if I send the same RA to all)

Thank you!

RobTillaart, 2020-12-27:
Thanks, 

It should be possible to use multiple of both - although I never tried it.

Please note that both RA and RM have their own repository now which has the latest version.
This repo will be kept sort of up to date.

https://github.com/RobTillaart/RunningAverage
https://github.com/RobTillaart/RunningMedian

Can you post a minimal sketch that shows the problem. _ e.g. 2RM's and 2 RA's
for the sensor just use the random() function than I can try to recreate your problem here 
(might take some time )

unstopp, 2020-12-27:
Right, it works if I do everything in one function. I was trying to do the following:

```
sensorAValue = sensorSaveData(sensorA, sensorASamples, sensorAMA); // Samples is RM value
sensorBValue = sensorSaveData(sensorB, sensorBSamples, sensorBMA);

long sensorSaveData(int sensor, RunningMedian sensorSamples, RunningAverage sensorMovingAverage) {
    sensorSamples.add(analogRead(sensor));
    //Serial.print("Running Median: ");
    //Serial.println(sensorSamples.getMedian());
    
    sensorMovingAverage.addValue(sensorSamples.getMedian());
    //Serial.print("Running Average: ");
    //Serial.println(sensorMovingAverage.getAverage());

    return sensorMovingAverage.getAverage();
}
```

But fails. If i throw everything inside a void sensorSaveData() (for both sensors) and print, it works.

RobTillaart, 2020-12-27:
Changed your post - added cpp to the code section (looks better)

You can make the RA/RM object global so there is no need to pass them to function.

unstopp, 2020-12-27:
I've updated the code. This is the setup I have

RobTillaart, 2020-12-27:
This is an example I quickly wrote,  It uses 5 RA's and 5 RM's that are also accessible  as array.

In setup() I first fill the 5 RM's with random numbers - to get started.
When 10 "measurements" are added the median is added to a RA

In loop() I add more random numbers to all 5 RM's and again after 10 I add it to the RA.
Then I print the average of all 5 RA's

Have a look at this code and see if this works for you,
You might swap the RA and the RM's of course.

```cpp
//
//    FILE: RunningBoth.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
// PURPOSE: demo
//    DATE: 2020-
//    (c) : MIT-
//

#include "RunningAverage.h"
#include "RunningMedian.h"

RunningAverage RA1(10), RA2(10), RA3(10), RA4(10), RA5(10);
RunningAverage RA[5] = { RA1, RA2, RA3, RA4, RA5 };

RunningMedian RM1(10), RM2(10), RM3(10), RM4(10), RM5(10);
RunningMedian RM[5] = { RM1, RM2, RM3, RM4, RM5 };

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.println();

  // fill all RM's
  for (int i = 0; i < 5; i++)
  {
    for (int a = 0; a < 10; a++)
    {
      RM[i].add(random(100));
    }
    RA[i].addValue(RM[i].getMedian());
  }
}

void loop()
{
  // ADD NEW VALUES
  for (int i = 0; i < 5; i++)
  {
    for (int a = 0; a < 10; a++)
    {
      RM[i].add(random(100));
    }
    RA[i].addValue(RM[i].getMedian());
  }

  // PRINT THEM
  for (int i = 0; i < 5; i++)
  {
     Serial.print(RA[i].getAverage());
     Serial.print('\t');
  }
  Serial.println();
}

// -- END OF FILE -- 
```

unstopp, 2020-12-27:
Thank you! I manage to get it working by adding everything in one function.
I was just trying to improve my code not to be crumblesome and use one general function for all 5 sensors.

I will try your method as well.

RobTillaart, 2020-12-27:
> I was just trying to improve my code, so it's not too crumblesome, and use one general function for all 5 sensors.

You can by passing the right index to the function.

unstopp, 2020-12-27:
Cool! Will try that, thank you! I think this can be closed, appreciate your help!

closed 2020-12-27 by RobTillaart