RobTillaart/Arduino #166: Weird avg values

ciberp, 2021-01-19:
Hi Rob!

I noticed that after a few days returned avg values are not right. I am using arduino 1.8.13 and ESP8266. Is it possible that I am hitting some overflow variable or something like that?

My code:
```
#include "Statistic.h"
Statistic AnalogValueStats(100); 
   
 void loop(void) {
      if (millis() - previousMillis >= 50) {
        previousMillis = millis();
        read_sensor_loop();
      }

void read_sensor_loop() {
      int AnalogValue = analogRead(A0);      
      AnalogValueStats.add(AnalogValue);
      AvgAnalogValue = AnalogValueStats.average();
      MaxAnalogValue = AnalogValueStats.maximum();
}
```
This is analogRead value:
[image]
And this is the returned drifting avg value, I think that avg is wrong calculated but I can't figured it out why?
[image]

Regards,
Peter

RobTillaart, 2021-01-19:
Hi Peter,

Strange, but never impossible.

Your code posted does not compile, please send a complete failing sketch (including all variable declarations.)

RobTillaart, 2021-01-19:
Question. 
What is the purpose of the 100 in the constructor?

ciberp, 2021-01-19:
> 
> 
> Question.
> What is the purpose of the 100 in the constructor? It expects a boolean.

Hmm, gut question :)

ciberp, 2021-01-19:
> 
> 
> Question.
> What is the purpose of the 100 in the constructor? It expects a boolean.
> 
> Question.
> Do you call `AnalogValueStats.clear()` in setup?
No

RobTillaart, 2021-01-19:
My runningAverage class has a parameter for size of an internal array,
The statistic class does not use one. The 100 is just interpreted as true so it should not do too much harm.

ciberp, 2021-01-19:
I hope this should compile since my whole skech is very large.
```cpp
#include "Statistic.h"

Statistic AnalogValueStats(100); 
int AvgAnalogValue = 0;
int MaxAnalogValue = 0;
unsigned long previousMillis = 0;
   
 void loop(void) {
      if (millis() - previousMillis >= 50) {
        previousMillis = millis();
        read_sensor_loop();
      }

void read_sensor_loop() {
      int AnalogValue = analogRead(A0);      
      AnalogValueStats.add(AnalogValue);
      AvgAnalogValue = AnalogValueStats.average();
      MaxAnalogValue = AnalogValueStats.maximum();
}
```

ciberp, 2021-01-19:
avg values graph since esp8266 reboot:
[image]
analogread values since esp8266 reboot:
[image]

It works great for a few days and than weird values...

RobTillaart, 2021-01-19:
There is no setup in the code.

ciberp, 2021-01-19:
I uploaded whole sketch https://github.com/ciberp/energy-meter

RobTillaart, 2021-01-19:
Better prepare a minimal sketch that shows the problem.

ciberp, 2021-01-19:
> 
> 
> Better prepare a minimal sketch that shows the problem.

Ok, thank you for quick response and your time. I will close this issue and when I am ready with minimal sketch that show the problem I will reopen the issue. I apologize for not preparing it right now.

Best Regards,
Peter

RobTillaart, 2021-01-19:
No need to apologize, but being prepared is always a sign it is a serious request.
Rob
(Netherlands, I expect to be online till 23:00)

ciberp, 2021-01-19:
Hi again!

I prepared the sketch and repeated the weird avg values.

arduino ver.: SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be

The sketch:
```cpp
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include "Statistic.h"

int AnalogValue = 0;
int AvgAnalogValue = 0;
int MaxAnalogValue = 0;
int n = 0;

Statistic AnalogValueStats(100);
//Statistic AnalogValueStats(100); 

unsigned long previousMicros = 0;
unsigned long previousMillisVM = 0;
unsigned long previousTerm = 0;

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin("ssid", "psk");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println(ESP.getFullVersion()); 
}

void loop() {
  if (micros() - previousMicros >= 50) {
    previousMicros = micros();
    yield();
    n++;
    if (n > 100) {
      AnalogValue = random(900, 980);
    } else {
      AnalogValue = random(810, 860);
    }
    if (n == 120) {
      n = 0;
    }
    AnalogValueStats.add(AnalogValue);
    AvgAnalogValue = AnalogValueStats.average();
    MaxAnalogValue = AnalogValueStats.maximum();
  }
  if (millis() - previousMillisVM >= 10000) {
    previousMillisVM = millis();
    yield();
    String VMDataStructure = "";
    VMDataStructure += "pulse,host=test,name=analog_avg value=" + String(AvgAnalogValue) + "\n";
    VMDataStructure += "pulse,host=test,name=analog_max value=" + String(MaxAnalogValue) + "\n";
    VMDataStructure += "pulse,host=test,name=analog value=" + String(AnalogValue) + "\n";
    send_to_vm_tcp(VMDataStructure);
  }
  if (millis() - previousTerm >= 5000) {
    previousTerm = millis();
    Serial.print("val, avg, max: ");
    Serial.print(AnalogValue);
    Serial.print(", ");
    Serial.print(AvgAnalogValue);
    Serial.print(", ");
[... 180 lines cut ...]
val, avg, max: 814, 865, 979
val, avg, max: 850, 861, 979
val, avg, max: 830, 858, 979
val, avg, max: 841, 854, 979
val, avg, max: 831, 851, 979
val, avg, max: 823, 848, 979
val, avg, max: 856, 845, 979
val, avg, max: 827, 841, 979
val, avg, max: 816, 838, 979
val, avg, max: 806, 835, 979
val, avg, max: 811, 832, 979
val, avg, max: 838, 829, 979
val, avg, max: 803, 825, 979
val, avg, max: 848, 822, 979
val, avg, max: 803, 819, 979
```

</details>

values graph:
[image]

Thank you for your in advance, I really appreciate.
Best regards,
Peter

RobTillaart, 2021-01-19:
MInimized your sketch to

```cpp

#include "Statistic.h"

int AnalogValue = 0;
int AvgAnalogValue = 0;
int MaxAnalogValue = 0;
int n = 0;

Statistic AnalogValueStats;
//Statistic AnalogValueStats(100);

unsigned long previousMicros = 0;
unsigned long previousMillisVM = 0;
unsigned long previousTerm = 0;

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  if (micros() - previousMicros >= 50)
  {
    previousMicros = micros();
    yield();
    n++;
    if (n > 100) {
      AnalogValue = random(900, 980);
    } else {
      AnalogValue = random(810, 860);
    }
    if (n == 120) n = 0;

    AnalogValueStats.add(AnalogValue);
    AvgAnalogValue = AnalogValueStats.average();
    MaxAnalogValue = AnalogValueStats.maximum();
  }

  Serial.print(AnalogValue);
  Serial.print(", ");
  Serial.print(AvgAnalogValue);
  Serial.print(", ");
  Serial.println(MaxAnalogValue);
}
```
when I look at it with the Serial plotter it behaves quite well

ciberp, 2021-01-19:
> 
> 
> MInimized your sketch to
> 
> ```c++
> #include "Statistic.h"
> 
> int AnalogValue = 0;
> int AvgAnalogValue = 0;
> int MaxAnalogValue = 0;
> int n = 0;
> 
> Statistic AnalogValueStats;
> //Statistic AnalogValueStats(100);
> 
> unsigned long previousMicros = 0;
> unsigned long previousMillisVM = 0;
> unsigned long previousTerm = 0;
> 
> void setup()
> {
>   Serial.begin(115200);
> }
> 
> void loop()
> {
>   if (micros() - previousMicros >= 50)
>   {
>     previousMicros = micros();
>     yield();
>     n++;
>     if (n > 100) {
>       AnalogValue = random(900, 980);
>     } else {
>       AnalogValue = random(810, 860);
>     }
>     if (n == 120) n = 0;
> 
>     AnalogValueStats.add(AnalogValue);
>     AvgAnalogValue = AnalogValueStats.average();
>     MaxAnalogValue = AnalogValueStats.maximum();
>   }
> 
>   Serial.print(AnalogValue);
>   Serial.print(", ");
>   Serial.print(AvgAnalogValue);
>   Serial.print(", ");
>   Serial.println(MaxAnalogValue);
> }
> ```
> 
> when I look at it with the Serial plotter it behaves quite well

Sorry I forgot to mentioned that it takes aboout 5min to get weird numbers, like it is shown in the graph...

RobTillaart, 2021-01-19:
think think think ....

Aha, yes some internal numbers overflow after some time.

The library has a variable **float sum**. Your sketch adds between 0 and 1000 every time. Given the precision of a float it is possible that after some 20.000 additions the next addition is not significant anymore.
However the other important variable **uint32_t count**, still increases (it can go to 4.000.000.000++), so we got in a scenario that sum stays constant and count increases.
That would cause the average = sum / count slowly degrade.

### TEST
Can you do a testrun and print  **AnalogValueStats.sum()**  and   **AnalogValueStats.count()** ?

### Version 0.4.2 ?
Do you have the latest version 0.4.2  -https://github.com/RobTillaart/Statistic ?
That one can still overflow but at least we are testing with the same version

As it is getting late I have no solution right now.
A first order band aid solution is replace all datatypes of float with double in the library. Then it will take far more time before the problem occurs.

to be continued tomorrow.

ciberp, 2021-01-20:
> think think think ....
>
> Aha, yes some internal numbers overflow after some time.
>
> The library has a variable **float sum**. Your sketch adds between 0 and 1000 every time. Given the precision of a [...]

I did statistic update from 0.4.1 to 0.4.2 from your link and fix the sketch like you suggested:

<details>

```cpp
#include "Statistic.h"

int AnalogValue = 0;
int n = 0;

Statistic AnalogValueStats(100);
//Statistic AnalogValueStats(100); 

unsigned long previousMicros = 0;
unsigned long previousMillisVM = 0;
unsigned long previousTerm = 0;

void setup() {
  Serial.begin(115200);
}

void loop() {
  if (micros() - previousMicros >= 20) {
    previousMicros = micros();
    yield();
    n++;
    if (n > 100) {
      AnalogValue = random(900, 980);
    } else {
      AnalogValue = random(810, 860);
    }
    if (n == 120) {
      n = 0;
    }
    AnalogValueStats.add(AnalogValue);
  }
  if (millis() - previousTerm >= 5000) {
    previousTerm = millis();
    Serial.print("val, avg, max, sum, count: ");
    Serial.print(AnalogValue);
    Serial.print(", ");
    Serial.print(AnalogValueStats.average());
    Serial.print(", ");
    Serial.print(AnalogValueStats.maximum());
    Serial.print(", ");
    Serial.print(AnalogValueStats.sum());
    Serial.print(", ");
    Serial.println(AnalogValueStats.count());
  }
 }
```

</details>

Yes, you are correct sum stays constant after 17179869184 :

<details>

```
val, avg, max, sum, count: 857, 964.78, 979.00, 16268537856.00, 16862481
val, avg, max, sum, count: 850, 965.20, 979.00, 16391465984.00, 16982528
val, avg, max, sum, count: 847, 965.61, 979.00, 16514776064.00, 17102948
val, avg, max, sum, count: 819, 966.02, 979.00, 16637747200.00, 17223037
val, avg, max, sum, count: 970, 966.42, 979.00, 16761069568.00, 17343469
val, avg, max, sum, count: 812, 966.82, 979.00, 16883889152.00, 17463410
val, avg, max, sum, count: 900, 967.21, 979.00, 17007192064.00, 17583823
val, avg, max, sum, count: 834, 967.59, 979.00, 17130012672.00, 17703765
val, avg, max, sum, count: 838, 963.86, 979.00, 17179869184.00, 17824120
val, avg, max, sum, count: 824, 957.40, 979.00, 17179869184.00, 17944278
val, avg, max, sum, count: 849, 951.01, 979.00, 17179869184.00, 18064802
val, avg, max, sum, count: 851, 944.77, 979.00, 17179869184.00, 18184204
val, avg, max, sum, count: 815, 938.55, 979.00, 17179869184.00, 18304706
val, avg, max, sum, count: 857, 932.43, 979.00, 17179869184.00, 18424821
val, avg, max, sum, count: 815, 926.37, 979.00, 17179869184.00, 18545351
val, avg, max, sum, count: 826, 920.41, 979.00, 17179869184.00, 18665402
val, avg, max, sum, count: 971, 914.51, 979.00, 17179869184.00, 18785862
val, avg, max, sum, count: 958, 908.72, 979.00, 17179869184.00, 18905509
val, avg, max, sum, count: 856, 902.97, 979.00, 17179869184.00, 19025944
val, avg, max, sum, count: 822, 897.31, 979.00, 17179869184.00, 19146031
val, avg, max, sum, count: 832, 891.70, 979.00, 17179869184.00, 19266485
val, avg, max, sum, count: 829, 886.18, 979.00, 17179869184.00, 19386450
val, avg, max, sum, count: 950, 880.70, 979.00, 17179869184.00, 19506955
val, avg, max, sum, count: 829, 875.35, 979.00, 17179869184.00, 19626269
val, avg, max, sum, count: 842, 870.01, 979.00, 17179869184.00, 19746688
val, avg, max, sum, count: 818, 864.76, 979.00, 17179869184.00, 19866603
val, avg, max, sum, count: 859, 859.55, 979.00, 17179869184.00, 19987093
```

[image]

</details>

I will also try your fix to change float with double and then lower samples (in my project) if i can managed to get correct avg numbers for a month and then I can do a reset aka AnalogValueStats.clear().

Thank you.

RobTillaart, 2021-01-20:
OK we found the cause, now to think of a good fix.

What I thought of last night was the following:

1. change the signature of add() to indicate effect of the addition. (good)
2. add a rescale function (bad & ugly)

```cpp
  // returns false if effectively nothing is added to sum.
  // either value == 0
  // or sum has reached "overflow status"
  bool add(const float value);

  // rescale multiplies sum * factor and count * factor  so average is conserved.
  // major drawback is that stddev etc is corrupted.
  void rescale(float factor);  //  0 .. 1
```

in your application I would use - https://github.com/RobTillaart/runningAverage

ciberp, 2021-01-20:
> 
> 
> OK we found the cause, now to think of a good fix.
> 
> What I thought of last night was the following:
> 
>     1. change the signature of add() to indicate effect of the addition. (good)
> 
>     2. add a rescale function (bad & ugly)
> 
>     3. double instead of float ??
> 
> 
> ```c++
>   // returns false if effectively nothing is added to sum.
>   // either value == 0
>   // or sum has reached "overflow status"
>   bool add(const float value);
> 
>   // rescale multiplies sum * factor and count * factor  so average is conserved.
>   // major drawback is that stddev etc is corrupted.
>   void rescale(float factor);  //  0 .. 1
> ```
> 
> in your application I would use - https://github.com/RobTillaart/runningAverage
> 
> by implementing (1) above there is at least an indicator of the problem.

Ok, will try with runningAverage. Thank you.

ciberp, 2021-01-20:
> OK we found the cause, now to think of a good fix.
>
> What I thought of last night was the following:
>
>     1. change the signature of add() to indicate effect of the addition. (good)
>
>     2. add a [...]

Maybe when overflow occurs clear should be made and this limitation is clearly stated. Like in my case other approach like running average should be used. Just my oppinion.

RobTillaart, 2021-01-20:
Valuable opinion, appreciated,

I already made the change to **add()**, so it returns a float so one can check the value added.

Also made an example that shows the problem in just a few seconds. 
See - https://github.com/RobTillaart/Statistic/tree/master/examples

There should also documentation in the README, maybe more explicit than this 
See - https://github.com/RobTillaart/Statistic#q-how-many-samples-can-the-lib-hold--internal-variables-and-overflow

The documentation should mention the **runningAverage** solution explicitly for applications that 
1. adds a lot of data. (like your scenario)
2. adds a lot of data of different orders of magnitude. (then one will miss values much faster)

**runningAverage** is a solution for (1) but not for (2) 

A solution for the second could be using logarithmic data?
The average of 1000 and 1 is 500.5 , in logarithmic terms that is log(1000) and log(1)  = 3 and 1  = ~2.7

And not all platforms support (8 byte) double so that isn't a (perfect) solution either..

For integer values I could use int64 which goes up to 10^18 something. That means 18 significant digits. 
That could be used to implement a fixed point float e.g. 12.6 digits (6 decimals) but whether the mapping is useful would completely depend on the numbers added... 

Lets start with documenting the problem.

RobTillaart, 2021-01-20:
documenting done
See - https://github.com/RobTillaart/Statistic#q-how-many-samples-can-the-lib-hold--part-2-order-of-magnitude-floats

ciberp, 2021-01-20:
> 
> 
> documenting done
> See - https://github.com/RobTillaart/Statistic#q-how-many-samples-can-the-lib-hold--part-2-order-of-magnitude-floats

Wow, very well explained.

RobTillaart, 2021-01-21:
Created a small library yesterday - infiniteAverage - to be published

it sums values in a float and if > 10000 it moves the tenthousands units to an uint32.

It ran the whole night on an UNO, added 12.6 million floats (0..1000) in ~8hrs = 30K seconds to it.
==> 400 samples added per second.

Here is a snapshot of the output

```
COUNT		OV		SUM		AVG
12645731	631546		3101.00		499.41
12645732	631546		3200.00		499.41
12645733	631546		3222.00		499.41
```

The overflow + sum combined  acts as a number with 9 + 7 = 16 digits sum
average = sum / count + 10000.0 * overflow / count.

in the beginning the "sum/count" is the most important part of the average, later as sum grows the second part becomes the largest contributor, however the first part still takes care of the precision.

There still is a limit in precision, but it is way better than the Statistics class. (to be investigated where the limits are)
Imagine uint64_t to count the whole part of the sum and the counter. (~18 digits) and a float for the decimal part (~7 digits).
that would allow years of sampling a billion samples per second. 

Done this experiment, I see the average becomes so stable that you won't notice outliers anymore.
This makes sense as every new sample add ```(sample - average)/count ``` to the new average.
As count grows and the ```sample - average``` stays roughly the same this added value slowly goes to zero.

(time for coffee)

RobTillaart, 2021-01-21:
changed the experiment, so sum would only hold the decimal part and that looks promising 

```
11199	5568	0.88196		0.4972661
11200	5569	0.33206		0.4972618
11201	5569	0.80366		0.4972595
```

RobTillaart, 2021-01-21:
@ciberp 
Think this issue can be closed.
If you need help with runningAverage, please open an issue on  https://github.com/RobTillaart/runningAverage

closed 2021-01-21 by RobTillaart

RobTillaart, 2021-01-21:
If you have time to play ;)

https://github.com/RobTillaart/infiniteAverage

ciberp, 2021-01-22:
> 
> 
> If you have time to play ;)
> 
> https://github.com/RobTillaart/infiniteAverage

I tested, added max function, and works perfect:

[image]

RobTillaart, 2021-01-22:
Looks good, but eventually it will have the same problem, but way later ...

Q: Did you implement the max as a float?
It is a good idea to add this maximum, minimum too

Q: what tool do you use for the graph's?

RobTillaart, 2021-01-22:
Doing a test on a ESP32, tight loop with random numbers => 365500+ numbers per second

Quick math => overflow after ~11.744 seconds = ~3h 15 minutes.

I'm thinking of embedding the following function in add

```cpp
void scale_back()
{
  if ((_count & 0x800000 > 0) ||     // _count hits 2 billion something  (maybe after 100 million?)
      (_whole & 0x8000000))          // _whole part hits 2 billion something
  {
   if (_whole & 1) _sum += 1;
    _ count /= 2;  
    _whole /=2;
    _sum /= 2;
  }
}
```

if one of the two hits the 2 billion it effectively drops half of the samples and half of the sum
=> average stays the same, 
=> the next value will still add only 1/billionth to the sum.
=> min and max will still be correct.
=> count will be corrupted / corrected.
=> whole and sum are corrupted /corrected.
=> as it scales back automagically it would be able to count forever...

Optionally I could add a 32 bit counter of the # corrections, that would give indication of real count again.
However using 64 bit would be programmatically easier.

ciberp, 2021-01-22:
> Looks good, but eventually it will have the same problem, but way later ...
> 
> Q: Did you implement the max as a float?
> It is a good idea to add this maximum, minimum too
Yes as float, oh I see you already added, thank you.
> 
> Q: what tool do you use for the graph's?
I use time series db [victoriametrics](https://github.com/VictoriaMetrics/VictoriaMetrics) and [grafana](https://grafana.com/) for graphing. I am running both on rpi4 with hard disk drive.

ciberp, 2021-01-22:
> Doing a test on a ESP32, tight loop with random numbers => 365500+ numbers per second
> 
> Quick math => overflow after ~11.744 seconds = ~3h 15 minutes.
> 
> I'm thinking of embedding the following function in add
> 
> ```c++
> void scale_back()
> {
>   if ((_count & 0x800000 > 0) ||     // _count hits 2 billion something  (maybe after 100 million?)
>       (_whole & 0x8000000))          // _whole part hits 2 billion something
>   {
>    if (_whole & 1) _sum += 1;
>     _ count /= 2;  
>     _whole /=2;
>     _sum /= 2;
>   }
> }
> ```
> 
> if one of the two hits the 2 billion it effectively drops half of the samples and half of the sum
> => average stays the same,
> => the next value will still add only 1/billionth to the sum.
> => min and max will still be correct.
> => count will be corrupted / corrected.
> => whole and sum are corrupted /corrected.
> => as it scales back automagically it would be able to count forever...
> 
> Optionally I could add a 32 bit counter of the # corrections, that would give indication of real count again.
> However using 64 bit would be programmatically easier.
This solution sounds very good, nice tricks :)

RobTillaart, 2021-01-23:
FYI I tried the trick of halving the sum and the count (and thought I did everything alright)
I compared the infiniteAverage with a separate counter + 8 byte double and up to 100.000.000 samples
the IA returned an average that had 7 decimals correct, after the **scale_back** it were "only" 5 decimals
that stayed correct. Not bad but definitely worse than I had expected. 
(That is exactly why one should investigate such tricks)

A second **scale_back** still got 5 decimals right but the magnitude of the error doubled.
A third **scale_back** got 4  decimals right 
A fourth **scale_back** got 4 decimals right 
A fifth **scale_back** still got 4 decimals right , magnitude error seems to stabilize

Test is calculating the average of random(10000) * 0.0001;  // random number between 0..1 ==> 0.5

Conclusion for now is that the trick works not as well as expected, and should be applied as late as possible, so indeed when
the internal uint32_t variables hit the most significant bit (2 billion ++)

RobTillaart, 2021-01-23:
```
MILLIS	_COUNT		_OVERFLOW	_SUM			average		reference	difference
3193	999725042	499810350	0.3810862303		0.4999478161	0.4999478150	0.0000000011
3194	1000037626	499966965	0.9899787307		0.4999481440	0.4999481549	0.0000000110
```

Ultimate testrun for scale_back just reached 1 billion, will takes ~hour before it triggers scale_back

_count = internal 32 bit counter
_overflow = integer part of sum
_sum = decimal part of sum
average = average (of course)
reference = (8 byte double / 64 bit counter) with same values
difference = abs(average - reference)

(should have added a column MAXdifference)

RobTillaart, 2021-01-23:
```
MILLIS	_COUNT		_OVERFLOW	_SUM			average		reference	difference
6872	2146904813	1073346919	0.1253623962		0.4999508560	0.4999508654	0.0000000094
6873	2147216555	1073502627	0.8103712201		0.4999507964	0.4999507969	0.0000000005
6874	1073786480	536840360	0.8025453687		0.4999507666	0.4999507294	0.0000000371
6875	1074097996	536995989	0.9052241445		0.4999506474	0.4999506766	0.0000000292
6876	1074409260	537151728	0.9562472701		0.4999507666	0.4999507336	0.0000000330

...

7913	1397210057	698539387	0.0586473942		0.4999530315	0.4999520136	0.0000010180
7914	1397521321	698694890	0.0046730042		0.4999529421	0.4999519674	0.0000009747
7915	1397832584	698850596	0.0775004625		0.4999530017	0.4999520036	0.0000009981
7916	1398143846	699006227	0.3528388143		0.4999530017	0.4999520098	0.0000009920
7917	1398455113	699162067	0.5848338604		0.4999531806	0.4999520994	0.0000010811
7918	1398766379	699317821	0.4931301475		0.4999532700	0.4999521544	0.0000011156
7919	1399077642	699473570	0.7307385802		0.4999533594	0.4999522080	0.0000011514
```

The scale-Back happend (from 2 billion to 1 billion) and the loss of precision is still 2 digits. 

Another strategy might be just drop 1% = would free up space for 20 million++ samples.

----

Using a double and an uint64_t is easier,  would take performance and 12 bytes of RAM as penalty.

RobTillaart, 2021-01-24:
After almost 24 hours   @307000 samples per second 

```
MILLIS	_COUNT		_OVERFLOW	_SUM			average		reference	difference
80086	2146629170	1073197526	0.9322826862		0.4999454618	0.4999492423	0.0000037806
80087	2146936691	1073351359	0.0707032681		0.4999455214	0.4999492459	0.0000037245
80088	2147244214	1073504929	0.4050865173		0.4999454319	0.4999492388	0.0000038068
80089	1073809911	536846412	0.5628843307		0.4999454916	0.4999492388	0.0000037473
80090	1074117434	537000127	0.3697922230		0.4999454618	0.4999492376	0.0000037758
80091	1074424955	537153904	0.9904112816		0.4999454916	0.4999492389	0.0000037473
```
80K x 307 K ~ 24.5 Billion samples and still 5 decimals right (in this test)

24.5 Billion samples means that the numbers are scaled back 23 times, every billion except the first 2.
So the scale back seems to have no accumulating effect. 

Promising result

RobTillaart, 2021-01-24:
> 24 hours still stable at 5 decimales  

```
90561	2146916567	1073345131	0.1331300735		0.4999473095	0.4999489258	0.0000016164
90562	2147224097	1073498915	0.6608264446		0.4999473095	0.4999489271	0.0000016176
90563	1073789795	536838449	0.1361019611		0.4999474287	0.4999489335	0.0000015048
90564	1074097319	536992279	0.4963036478		0.4999475181	0.4999489365	0.0000014184
```