RobTillaart/Arduino #168: Reading and printing uint16t

EliasA97, 2021-02-26:
Hello, your fram library is amazing. It helped me a lot with my project.. I have a question though. When I store a uint16t or 32 number (let's say 65000 or 1200000), I can successfuly read it back but without the points. How can I make this 65000 to this 65.000. Also this 1200000 to this 1.200.000?

RobTillaart, 2021-02-26:
Will look into this tomorrow.

RobTillaart, 2021-02-27:
@EliasA97 
The issue is not a problem of the FRAM library (which has its own repo here =>  https://github.com/RobTillaart/FRAM_I2C

A number is a number and we humans give it a representation that is easy for us. The computer just has bytes and bits and the instruction to handle those bits as an int or float string whatever.  We humans prefer decimal notation or sometimes hexadecimal of formatted as a currency, percentage or ...

What you are after is a function that converts a number into a formatted string / char array that has a thousand separator, a decimal separator and possibly even a currency symbol. (as currency is typical the place where you use .'s in numbers)

```char currency(int32_t value, uint8_t decimals, char symbol, char tsep, char dsep);```

This is not a trivial function to write quickly as there are multiple corner cases to consider. 

probably you need to start with a the minimalistic version that solves your question.
```char currency(int32_t value, char tsep);```

usage
```
Serial.println(currency(value, '.');
Serial.println(currency(value, ',');
```

Note there exists different notation systems with dots e.g. - https://en.wikipedia.org/wiki/Indian_numbering_system

RobTillaart, 2021-02-27:
first code to get you started, 
does not handle negative numbers

```cpp
//
//    FILE: currency.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
// PURPOSE: display an int as currency
//    DATE: 2021-02-28
//     URL: 

#include "Arduino.h"

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);

  Serial.println( currency(0, '.') );
  Serial.println( currency(9, '.') );
  Serial.println( currency(10, '.') );
  Serial.println( currency(99, '.') );
  Serial.println( currency(100, '.') );
  Serial.println( currency(999, '.') );
  Serial.println( currency(1000, '.') );
  Serial.println( currency(9999, '.') );
  Serial.println( currency(10000, '.') );
  Serial.println( currency(99999, '.') );
  Serial.println( currency(100000, '.') );
  Serial.println( currency(999999, '.') );
  Serial.println( currency(1000000, '.') );
  Serial.println( currency(9999999, '.') );
  Serial.println( currency(10000000, '.') );
  Serial.println( currency(99999999, '.') );
  Serial.println( currency(100000000, '.') );
  Serial.println( currency(999999999, '.') );
  Serial.println( currency(1000000000, '.') );

  Serial.println("done");
}

void loop()
{
}

char * currency(uint32_t value, char tsep)
{
  static char tmp[24];
  sprintf(tmp, "%ld", value);
  int len = strlen(tmp);
  int dots = (len - 1) / 3;

  tmp[len + dots] = 0;
  for (int i = len-1; i > 0; i -= 3)
  {
    tmp[i + dots - 0] = tmp[i];
    tmp[i + dots - 1] = tmp[i - 1];
    tmp[i + dots - 2] = tmp[i - 2];
    tmp[i + dots - 3] = '.';
    dots--;
  }
  return tmp;
}

// -- END OF FILE --
```

closed 2021-02-27 by RobTillaart

EliasA97, 2021-02-27:
WOW. works like magic. It's 2nd time you help me with this library and I must tell that thanking you is not enough!!!

RobTillaart, 2021-03-01:
related - https://github.com/RobTillaart/Currency