RobTillaart/Arduino #170: AD524X

Manhem, 2021-04-30:
Hi. 
I just wonder if the AD524X library for Arduino also works for AD5144?

RobTillaart, 2021-04-30:
Hi,
Never worked with the AD5144, I would need to check the datasheet to see if there is a (partial) match. 
or you can just give it a try...

RobTillaart, 2021-04-30:
Better discuss topic at its main repo - https://github.com/RobTillaart/AD524X

RobTillaart, 2021-04-30:
A quick look shows the AD5144 has 4 potmeters so best case it will work for only two of them...
And more important the AD5144 is a SPI device so that will not work with this I2C based lib.
Sorry...

The AD5144A is an I2C variant of the chip.

https://www.analog.com/media/en/technical-documentation/data-sheets/AD5124_5144_5144A.pdf

Based upon this lib it might not be too difficult to derive an AD5144A library.

Manhem, 2021-04-30:
Sorry.
I forgot the A at the end.
It's a AD5144A.
So if you have time and want to help me, can you see if you can implementate the AD5144A in you library?
I will be very greatful if you can help me.

RobTillaart, 2021-04-30:
Created an empty repo - https://github.com/RobTillaart/AD5144A - to develop a library
I cannot promise any timescale but I have some time this weekend so lets see how far we can go.
As I do not have any hardware I rely on your testing ...

Please reopen this issue there, please add a link to this one (so this issue can be closed.

closed 2021-04-30 by RobTillaart