RobTillaart/Arduino #178: Load cell liquid filling

Oceanlig, 2021-12-12:


RobTillaart, 2021-12-12:
Can you give a full description of the issue ?

Oceanlig, 2021-12-12:
I have a edible oil filling/bottling machine installed. It costs 1000$. It puts oil as per prescribed qty. And stops when the weight of bottle reaches say 1000 gram.
I want to make that machine myself. I think it would be way too cheap. I have an accounting background. And don't know anything about it. But I think I can do it with some help

RobTillaart, 2021-12-12:
(thinking out loud)
There are two distinct ways to tackle this. 
- you can weigh every bottle - which should gain 1000 grams
- you can weigh the bulk supply - which should drop 1000 grams 

The advantage of method 1 is that you can use relative small load cells e.g. 3x 500gr and measure it quite precise.
Three load cells could make a stable triangle construction.  
500gr LC have a resolution of about 1 gr so that would be good, but note that they should also weigh the bottle and possibly part of the construction so maybe 4 x 500gr LC is better as the load cells are not at the end of their operational range. (good for long term stability)

Assuming the bottles are uniform in weight you could use the **tare()** function or better weigh each bottle when empty and use that as starting point. This latter method allows you to detect bottles that are outside the expected bottle weight e.g. because there is some pollution, or thinner glass whatever.

Weighing the supply is possible simpler in construction, but I expect less precise as (assumption) the bulk weighs 30Kg so it is difficult to reach the 1 gram accuracy. 

Q1: I want to move this issue to https://github.com/RobTillaart/HX711 
OK with you?

Q2: what kind of help do you expect from me?
As this is not a "bugfix" request in the library.

Oceanlig, 2021-12-13:
If I use 4 500gr load cell. How much capacity can I fill. I mean upto how much weight. Like if I need to fill 10kg of liquid, then will this hold?

Oceanlig, 2021-12-13:
Q2. I am new to GitHub. So I don't know how to use it proper. You can move the issue to proper place. 
Q3. I live in india.

RobTillaart, 2021-12-13:
> If I use 4 500gr load cell. How much capacity can I fill. I mean upto how much weight. Like if I need to fill 10kg of liquid, then will this hold?

No, the devices will be overloaded. I propose you do some reading how load cells work in detail. About working range precision etc.  Think Adafruit has a good backgrounder. But many more exist.

GitHub is not the place to learn this. This library is about controlling the amplifier for the signals of load cells in software. GitHub is the place to discuss bugs or new features in that software. 
A better place might be the Arduino Forum.